bambulab/Bambu-Handy #185: Feature Request: Add button to disable toolhead light

kiler129, 2024-03-22:
Leaving the printer at night one can disable the usual light bar, as long as timelapse is disabled. However, the toolhead logo light is exceptionally bright when looked at directly.

It would be handy (hah!) to get a button that disables not only the lamp, but also the logo backlight. I know it's possible via GCode, but one cannot easily execute it mid-print. Also, the toolhead light should probably turn off after the print.

gschintgen, 2024-11-21:
Related: It should also be possible to manually turn the light *on*. Currently I can check the video feed of a remote printer but it's impossible to see whether the print bed is clear or not since there's no light, neither in the room nor in the printer. What good is a video feed if it's only showing black darkness?

gschintgen, 2024-11-21:
Sorry for the noise, i.e. my previous post: The option to manually switch the toolhead light is already there. I just have to tap "Printer" and at the very bottom there is a button for it. (The button is slightly overlapped by Android's bottom toolbar. Minor UI issue.)