bambulab/Bambu-Handy #186: Feature request: Live Activities for monitoring prints on iOS

luisillo26, 2024-03-23:
It would be great that you could implement iOS [Live Activities](https://developer.apple.com/design/human-interface-guidelines/live-activities) for monitoring and controlling Bambu Handy while a print is in progress.

Similar to what MobileRaker does with Klipper printers and OctoPod for Octoprint

Mike61704, 2024-05-25:
Would be good to have this on Android as well in the form of a widget.

nathantsmith, 2024-07-12:
I additionally came here just to say the same.  I was going to consider a pull request, but saw that handy was unfortunately not open source. This would be a great addition for iOS.  Just make sure it is also compatible with iOS's Standby Mode!

nathantsmith, 2024-07-12:
I will offer my services to implement this for the team :) Give me an NDA and I'll do it in my spare time for a second AMS

nramo013, 2024-09-02:
Agreed. This would be very handy to help with time and resource management as well. Especially for those with multiple printers going off

seanmcne, 2024-11-04:
I'm a user that's new to Bambu and I've previously used octoprint - overall the experience has been good but I miss the live activity on iOS so I can see the status and also quickly jump into a live print video stream.

StevenLi-phoenix, 2025-02-23:
我非常支持在 Bambu Handy 中引入实时进度监控功能。这将使用户能够直接在 iOS 设备上查看打印进度，提升使用体验。类似的功能在其他平台上已被证明有效，例如 MobileRaker 和 OctoPod。希望 Bambu Lab 能考虑将此功能添加到应用中。

StevenLi-phoenix, 2025-02-23:
Found one old one here: #36

StevenLi-phoenix, 2025-03-02:
Should be closed now @luisillo26 
https://apps.apple.com/us/app/bambu-handy/id1625671285
In update 2.19.0
```
 Added support for Live Activities to show print progress on ios
- 增加了对 Live Activities 的支持，以在 iOS 上显示打印进度
```

luisillo26, 2025-03-25:
Thanks for making this possible, it works great!

closed 2025-03-25 by luisillo26

user858753257, 2025-04-13:
How to start a live activity ?

StevenLi-phoenix, 2025-04-13:
Just go to devices page when printing.And then you could lock your phone/back to Home Screen.Live view will appear similar to uber.来自李树雨的iPhone