bambulab/Bambu-Handy #193: Exclude Object desactivated with cache file

MaximeBerduck, 2024-04-21:
I can't exclude object when printing from a printer cache (X1C, P1S, A1 mini).

It tells me: "Object skipping is not supported in the following cases: 
-more than 64 objects in a single plate 
-Bambu Studio slicer lower than V1.6".

Whereas I have 2 objects on the printout and the latest stable version of Bambu Studio (1.8.4.51)

TimWoosh, 2024-05-19:
I'm having the same issue with the latest Bamby handy app and Bambu Studio v1.9.1.66.

Bambu Lab A1 with AMS Lite
6 objects on the build plate.

Nakonechnyi, 2024-05-27:
It looks like an issue of a printer's firmware, as the Bambu Studio also can't see the model's preview. 
+ Print cache is a 3mf file (from a display interface) and app doesn't see the preview and it can't define objects. But if you look into the SD card files, prints have other two files gcode and some other (forgot which one). Either the files don't cache enough or the program can't reconstruct it.