bambulab/Bambu-Handy #200: No notifications

NinoHelfenstein, 2024-05-27:
I don't get any notifications on my phone running android. Everything I could think of is turned on but still no notifications.

LibaoBambu, 2024-05-30:
Thank you for your feedback. Can you provide more information: https://bambulab.com/en-eu/my/support/tickets

TheDutchDev, 2024-06-18:
Been having the same issue for a week or so now. It used to work fine, didnt change anything, now its broken.

No print complete, no errors, nothing. I have to manually open the app to see whats going on.

VladimirVecera, 2024-10-13:
I have the same thing. No notifications at all. And if I want to turn off the support ticket, I get an error: "Application error: a client-side exception has occurred (see the browser console for more information)."

chadheise, 2025-01-26:
I do not get any notifications despite having them enabled in my app settings and phone settings. Makes it very hard to monitor prints and I’ve wasted lots of time not realizing a print stopped due to an error or running out of filament. Please fix.

WindArrow3d, 2025-03-27:
I finally got Bambus because the entire world has been screaming that they're perfect and flawless for years now. I've printed with 6 P1S combos for a week and have gotten exactly zero notifications related to the printers. All notification settings are turned on in the app and in my phone settings. I just don't understand. Why doesn't this work? Why are people claiming it does when it clearly doesn't?