bambulab/Bambu-Handy #204: Colors not showing up correctly on Bambu Handy

sdphadke, 2024-07-15:
Issue:
When I try to select colors for a print, they don't show up correctly in the app. I can't use my AMS Lite from mobile because of this. Even on Bambu filaments, they just show up as a bunch of random colors. This happened with Bambu PLA Basic and PLA Matte. I don't have any other materials so I am not sure if this is a widespread problem.

Opinion:
I can't believe that they haven't kept it updated with their own portfolio.

pepsiliu, 2024-07-16:
Thanks for your ask. I think we need more information about this issue. 
1. where is the printing start from?
2. where and how did you make the colors selection?
3. did it show up the random colors at the printing page or device page?
Please provide a screenshot or video to help us to reproduce this issue .

sdphadke, 2024-07-18:
https://share.icloud.com/photos/061alkNmTgQ1UNHwUKQLKIaXw
Above is the link for the recording that I took. Ignore the printer connection failed, my printer is turned off. The issue persists even when it is on.

pepsiliu, 2024-07-19:
> https://share.icloud.com/photos/061alkNmTgQ1UNHwUKQLKIaXw Above is the link for the recording that I took. Ignore the printer connection failed, my printer is turned off. The issue persists even when it is on.

@sdphadke The colors on the 3D-Preview are preset. And It also will display you AMS/AMS lite's filament colors if your AMS
works well, the external spool was not displayed yet.