bambulab/Bambu-Handy #208: add iOS Siri commands

notAnElephant, 2024-07-21:
It would be cool if the app's most basic features were integrated with Siri, so e.g. you could ask it to tell you how much time is left from your print, or what is the current nozzle temperature, etc.

bobobox, 2024-09-24:
This would be nice. Another use case is integration with Apple Shortcuts - I have the printer on a smart switch and can say "Hey Siri, turn off the 3D printer", but what would be even better is if I could add a condition that only does that if there's no job printing.

Jelucan, 2025-05-08:
Please give fonction to iOS short ut app for automate Things with the printing state. Like when print is done turn on or off a light