bambulab/Bambu-Handy #219: Bambu Handy 2.15.0 cannot connect to internet

Merlwynd, 2024-10-15:
Updated Bambu Handy to new 2.15.0 version on my iPad Air 2 running ios 15.8.3. Now Bambu Handy says failed to connect to internet. Can’t do anything with the app from here as cannot get to account or Makerworld or support etc. Deleted and redownloaded app and same result. No issues before new version of app. Also no problems running the new app on my iPhone or connecting to printer via Bambu Studio on my pc, so my network is fine….

ophirf15, 2024-10-17:
Having the same issue here too.

Merlwynd, 2024-10-17:
Have submitted a ticket to BambuLab. They asked a couple questions but no solid recommendations forthcoming yet…..

SkyCatcher2, 2024-11-10:
Incase this is still an issue I figured out the solution after encountering the same error.

my iPhone name had special characters “Name’s IPhone” which caused an exception in the app. After I renamed the iPhone to “Names_Iphone” the app worked again and I could log in and connect.

Merlwynd, 2024-11-11:
Thanks....in my case a simple network reset solved the issue....but your solution may help others....

closed 2024-11-11 by Merlwynd