bambulab/Bambu-Handy #223: Cannot log into bambu handy

btcmechanic, 2024-11-04:
Bambu Handy
2.15.1 (android)

I get error message:

"Your account has been detected in another device. To keep it secure, please update the app and login again."

splaisan, 2024-11-04:
Thanks for posting here, I did not find this repo when I reported the issue to [bambu studio](https://github.com/bambulab/BambuStudio/issues/5142) more appropriate indeed.

mncasper78, 2024-11-05:
Same issue.   Signed up for Bambu Studio without issue on PC.  Then downloaded Handy on my Android and received the error.
Date 11/4/24
Time Approx 7pm America CST
Windows 11 Pro for PC
Samsung S23, SMS911U, Android 14

splaisan, 2024-11-05:
apparently the issue was fixed since yesterday's post, I can now access the app after obtaining and entering the security  code.