bambulab/Bambu-Handy #225: TimeoutException after 0:00:05.000000: start tutk server timeout

MichMich, 2024-11-10:
While trying to start the video stream in Bambu Handy, I get the following error: 

```
TimeoutException after 0:00:05.000000: start tutk server timeout
```

I'm running the latest firmware on my X1C and iOS 18.0.1 on my iPhone 15 Pro. 

Please advise on how to debug.

gschintgen, 2024-11-19:
I had that too on my Android phone today. It happened at the start or the print (bed leveling or some other calibration routine running on the X1C). A few minutes later the connection was fine.

The print progress was shown, as far as I could tell only the video feed was affected.

firemanz8, 2024-12-06:
Any fix for this, I get it whenever I try to stream the video

akorczak, 2024-12-09:
Same for me. Video stopped working a week ago or so. Pixel 8 Android 15

JPinPA, 2024-12-20:
Happened to me last night and still an issue today.[image]

swpaxton66, 2024-12-24:
Same issue here.  I have an Android, video feed will not start with the same tukt server error message.  I also cannot connect to the video feed on the windows desktop Bambulab Studio. Seems like this is an update or bambulab system / server issue.

swpaxton66, 2024-12-26:
FYI.  I am not sure what the root cause was.  But now my camera feed is working on both the Android BambuLab Handy app and my desktop.  What I did was clear the SD card, turn off LAN preview (which was always on and the camera was working but I was not in LAN mode) and hard power cycled the printer.  Not sure if my actions fixed my problem or BambuLab made a system change.

Numbski, 2025-01-02:
It temporarily resolves if you restart the printer.  I have had to reboot my printer after nearly every print job for months because of this.  Went looking for a better solution and ended up here.  tutk is a cloud video streaming product, so it seems to me that the tutk_server process on the printer itself is crashing.

JPinPA, 2025-01-03:
> It temporarily resolves if you restart the printer.  I have had to reboot my printer after nearly every print job for months because of this.  Went looking for a better solution and ended up here. [...]

I ultimately disassociated the printer and then reregistered it. No issues so far since.

Numbski, 2025-01-03:
I wish.  I just did that a few days ago to get custom filaments to populate on the printer.  Those have vanished again though, so I will likely do it again soon.

Tony Shadwick

Numbski, 2025-01-05:
Have since logged the device out and back in.  Still have to reboot after every print.