bambulab/Bambu-Handy #233: Hope Bambu-Handy add the NFC reading function

ShanLangDang, 2025-02-11:
希望app端增加NFC读取耗材的功能，原厂耗材装有RFID标签，手机NFC可读取。目前颜色种类太多，有时候近似颜色，近似材质的不好区分，只能放进机器识别，因此建议在APP端增加NFC功能，方便查看耗材颜色、材质、预估数量。
I hope the app can add NFC reading function for consumables. The original consumables are equipped with RFID tags that can be read by NFC on mobile phones. At present, there are too many types of colors, and sometimes it is difficult to distinguish between similar colors and materials. They can only be recognized by machines. Therefore, it is recommended to add NFC function on the APP to facilitate the viewing of consumable colors, materials, and estimated quantities.

nce8198, 2025-02-11:
善

Petersburg16, 2025-08-22:
善

m0h31h31, 2026-01-17:
https://github.com/m0h31h31/BambuRfidReader/releases/tag/v1.0  希望能帮到你