bambulab/Bambu-Handy #234: Failed to get ams mapping, is back?

TamiRose, 2025-02-28:
I printed one multicolor print from the sd card (the original one that came with the P1S) and it printed fine. I tried another, and it got to the last color change and I got the failed to get AMS mapping. The retry button did nothing, had to cancel the print. 

[image]

LibaoBambu, 2025-02-28:
Please create a ticket to provide more info [https://bambulab.com/en-eu/my/support/tickets](url), Thanks.

TamiRose, 2025-02-28:
I cant seem to upload lof Giles, so I dont know what too do since it won't let me open a support ticket without that

LibaoBambu, 2025-02-28:
Hope it helps：https://wiki.bambulab.com/en/X1/troubleshooting/how-to-upload-log

TamiRose, 2025-02-28:
I just went and did it from the handy app, that way worked. thank you

closed 2025-02-28 by LibaoBambu