bambulab/Bambu-Handy #242: Crash on launch with v3.1.0

grrrrr, 2025-04-29:
- **OS:** Android 15
- **Security Patch**: 2025-04-05
- **Handy version:** 3.1.0 (5020)

Following an upgrade to the latest version of Babmu Handy, the app now crashes when you try to launch it.

```
type: crash
osVersion: google/akita/akita:15/BP1A.250405.007.B1/2025042500:user/release-keys
userType: full.secondary
package: bbl.intl.bambulab.com:5020, targetSdk 34
process: bbl.intl.bambulab.com
processUptime: 279 + 271 ms
installer: com.aurora.store

java.lang.UnsatisfiedLinkError: JNI_ERR returned from JNI_OnLoad in "/data/user/10/bbl.intl.bambulab.com/.jiagu/libjgbibc_64.so"
	at java.lang.Runtime.load0(Runtime.java:964)
	at java.lang.System.load(System.java:1729)
	at moc.balubmab.ltni.lbb.a.a(SourceFile:233)
	at moc.balubmab.ltni.lbb.StubApp.attachBaseContext(SourceFile:188)
	at android.app.Application.attach(Application.java:352)
	at android.app.Instrumentation.newApplication(Instrumentation.java:1368)
	at android.app.LoadedApk.makeApplicationInner(LoadedApk.java:1473)
	at android.app.LoadedApk.makeApplicationInner(LoadedApk.java:1405)
	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:7697)
	at android.app.ActivityThread.-$$Nest$mhandleBindApplication(Unknown Source:0)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2509)
	at android.os.Handler.dispatchMessage(Handler.java:109)
	at android.os.Looper.loopOnce(Looper.java:232)
	at android.os.Looper.loop(Looper.java:317)
	at android.app.ActivityThread.main(ActivityThread.java:8973)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:591)
	at com.android.internal.os.ExecInit.main(ExecInit.java:50)
	at com.android.internal.os.RuntimeInit.nativeFinishInit(Native Method)
	at com.android.internal.os.RuntimeInit.main(RuntimeInit.java:369)
```

nocluewhatido, 2025-04-29:
I got the same Problem running GrapheneOS

su-sd, 2025-05-02:
I have the same problem, tried to allow all sorts of permissions.

AdamBath, 2025-05-03:
> I got the same Problem running GrapheneOS

Same on Graphene on my Pixel 7

mike-lloyd03, 2025-05-03:
Having the same issue on Graphene as well with "Exploit protection compatibility mode" enabled.

su-sd, 2025-05-04:
Has anyone this problem and is not on GrapheneOS?

nocluewhatido, 2025-05-05:
I recently installed Bambu Handy Version 3.0.3 (5010) using APK Mirror. This version functions correctly with the "exploit protection compatibility mode" enabled.

I reached out to Bambu Lab app support regarding this issue but have not received a response yet.

During my research, I came across a Reddit post that discusses a potential reason for the issue. The post suggests that Bambu Handy employs a protector for their app, which is developed by the Chinese security company 360.

You can find the Reddit post here: https://www.reddit.com/r/OpenBambu/comments/1i863hn/comment/mpnsuvu/?context=3

GitEva1, 2025-05-06:
Same problem. Downgrade to version 3.0.3 (5010) works fine.

grrrrr, 2025-05-08:
linking this dead [thread](https://discuss.grapheneos.org/d/22220-bambu-handy-app-crashing-on-start-up) I came across with same issue  over on GrapheneOS forums

GitNitneroc, 2025-05-13:
Still a problem today

LibaoBambu, 2025-05-13:
Please download **3.1.1 version** in  Google Play to see if the issue is sorted out now. Let us know if still having trouble! Thanks.

GitNitneroc, 2025-05-13:
I only see 3.1.0 in google play... Is this a beta version or something I should enable first ?

LibaoBambu, 2025-05-13:
Store might need some time to synchronize.

nocluewhatido, 2025-05-13:
> Please download **3.1.1 version** in  Google Play to see if the issue is sorted out now. Let us know if still having trouble! Thanks.
> 

Now Bambu Handy (3.1.1) is working on GrapheneOS even with exploit protection compatibility mode disabled.

Many thanks for the fix!