bambulab/Bambu-Handy #244: App no longer opens on Android

mike-lloyd03, 2025-05-03:
Recently, I cannot open the app on my Android phone. It immediately crashes. I've tried clearing the cache and reinstalling and still no luck. Here's the log from the crash:

```
type: crash
osVersion: google/shiba/shiba:15/BP1A.250405.007.B1/2025042500:user/release-keys
flags: dev options enabled
package: bbl.intl.bambulab.com:5020, targetSdk 34
process: bbl.intl.bambulab.com
processUptime: 253 + 60 ms
installer: com.android.vending

java.lang.UnsatisfiedLinkError: JNI_ERR returned from JNI_OnLoad in "/data/data/bbl.intl.bambulab.com/.jiagu/libjgbibc_64.so"
	at java.lang.Runtime.load0(Runtime.java:964)
	at java.lang.System.load(System.java:1729)
	at moc.balubmab.ltni.lbb.a.a(SourceFile:233)
	at moc.balubmab.ltni.lbb.StubApp.attachBaseContext(SourceFile:188)
	at android.app.Application.attach(Application.java:352)
	at android.app.Instrumentation.newApplication(Instrumentation.java:1368)
	at android.app.LoadedApk.makeApplicationInner(LoadedApk.java:1473)
	at android.app.LoadedApk.makeApplicationInner(LoadedApk.java:1405)
	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:7697)
	at android.app.ActivityThread.-$$Nest$mhandleBindApplication(Unknown Source:0)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2509)
	at android.os.Handler.dispatchMessage(Handler.java:109)
	at android.os.Looper.loopOnce(Looper.java:232)
	at android.os.Looper.loop(Looper.java:317)
	at android.app.ActivityThread.main(ActivityThread.java:8973)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:591)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:927)
```

grrrrr, 2025-05-03:
this looks to be a duplicate of #242

mike-lloyd03, 2025-05-03:
You're right. Thank you. Closing.

closed 2025-05-03 by mike-lloyd03